ghi-electronics/TinyCLR-Ports #6: Question: build.bat refactoring

kodfilemon, 2017-07-14:
It is possible to refactoring build.bat?
Maybe better way (even temporary) is move set parameters from build.bat to every Devices port.
For example in Devices\FEZ directory create args.txt file with content:
ProcessorPart=STM32F4
ProcessorArchitecture=CortexM4
ImageGenParameters=0x884DED08 0x3671259A 0x08008000 0x00038000

And in build.bat in place of IF ELSE tree section:

SET ARGSFILE=%ScriptRoot%\Devices\%Device%\args.txt

IF EXIST "%ARGSFILE%" (
  for /f "delims=" %%A in ('type "%ARGSFILE%"') do (
      SET %%A
  ) 
) ELSE (
  ECHO Unsupported device passed: %Device%
  GOTO :EOF
)
In this way is simplify create new port (copy) without touching build.bat.

Arke64, 2017-07-14:
There's still a lot of work needed in build.bat, but this would be a quick and easy change that we're open to.

Arke64, 2017-08-03:
Done. It's actually been split up into three different files now, one each for the device, target, and core.

commit f0bf0ec8 (2017-08-03): Moved device, target, and core specific configuration to separate files, fixes #6.
closed 2017-08-03 by Arke64