ghi-electronics/TinyCLR-Ports #26: Microseconds for .NET

valoni, 2017-08-29:
**Just for IDEA** 

- did is possible to create 2 Microseconds Method API to be used from .NET 
which could be executed as separated thread (or any better solution for cases to deal with Microseconds Needs):

1.)
void MicrosecondsForOneGpio(int HighTime,int LowTime,int Repeat,gpio Pin)
{ 
   for(int i=0;i<=Repeat;i++)
   {
      pin.Write(High);
      wait_microseconds(HighTime)
      pin.Write(Low);
      wait_microseconds(LowTime)
   }
}

2,)
void MicrosecondsForTwoGpio(int Gpio1Time,int Gpio2Time,int pause1, int pause2,gpio Pin1, gpio Pin2, int Repeat)
{ 
   for(int i=0;i<=Repeat;i++)
   {
      pin1.Write(High);
      wait_microseconds(Gpio1Time)
      pin1.Write(Low);
      if(pause1>0) wait_microseconds(pause1)
      pint2.Write(High);
      wait_microseconds(Gpio2Time);
      pint2(Low);
      if(pause2>0) wait_microseconds(pause2);
   }
}

I belive those will be enought in case when we need apply parts with microseconds without needs for Marshalling (PE / ex-RLP files).

Arke64, 2017-08-29:
For your first example, does GpioChangeWriter/SignalGenerator take care of it? http://docs.ghielectronics.com/tinyclr/tutorials/gpiochangewriter.html has some details and points to the original doc.

Arke64, 2017-12-07:
Closed for no response.

closed 2017-12-07 by Arke64