ghi-electronics/TinyCLR-Ports #50: STM32F4 targets break if i modify some parameters in DeviceSelector.h

valoni, 2017-09-08:
**i tried different scenarios for Nucleo STM32F411RET6**

1.)i can not add SPI more than 2 **(if i add it break compile)**

2.)i can not add UART more than 2 (if i add have problem with defination)
 tried diffeent combination but do not function (even with GPIO_PIN_NONE) 

3.)i need to comment  #define INCLUDE_USBCLIENT to be able 
to compile with defination  #define UART_DEBUGGER_INDEX 1

4.)and last one i can not compile with    #define STM32F411xE                     1 
    but work if I use #define STM32F401xE                     1

Arke64, 2017-09-11:
Can you create separate issues for both 3 and 4?

I was able to add an additional UART and SPI to the G80 port. I just added one set of pins to respective pins definition and increased TOTAL_*_CONTROLLERS by one.

valoni, 2017-09-11:
ignore issue 3 now work well since i explain i have problem 
and in case for SPI and UART i will take a look how you added it

 just issue 4 is problematic

Arke64, 2017-09-11:
To improve debugger selection, take a lot at #53. For issue 4, please create a new issue describing the issue and listing the errors you see. I'll be closing this issue then.

closed 2017-09-11 by Arke64