ghi-electronics/TinyCLR-Ports #169: Building a string longer than 127 characters with variable concatenation corrupts the string.

rickmas, 2018-01-31:
TinyCLR 0.7.0

A string variable assigned more than 127 characters through concatenation that includes at least one variable  gets corrupted.

For example this code:
```c#
            //int v = 123;
            string v = "123";
            //                     1         2         3         4         5         6         7         8         9         0         1         2
            String s0 = $"12345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678";

            String s1 = $"{v}4567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567";     //Interpolated String
            String s2 = $"{v}45678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678";    //Interpolated String

            String s3 = v + "4567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567";     // + concatenation
            String s4 = v + "45678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678";    // + concatenation

            String s5 = String.Format("{0}4567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567", v);    //String.Format
            String s6 = String.Format("{0}45678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678", v);   //String.Format
```
shows:
* s0 with a length longer than 127 characters but no concatenation.
* s1, s3, s5 demonstrate different methods for concatenation resulting in strings that are 127 characters and have **no** issue.
* s2, s4, s6 each have one additional character (128) and all are corrupt.  Interestingly though the Length property returns the proper size of 128 even though the string appears corrupted.

Here's a table showing the results:

Name | Concatenation Type | Length | Issue? | Resulting Value
-- | -- | -- | -- | --
s0 | None | 128 | No | "12345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678"
s1 | Interpolation | 127 | No | "1234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567"
**s2** | Interpolation | 128 | **Yes** | "@l���"
s3 | + | 127 | No | "1234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567"
**s4** | + | 128 | **Yes** | "�b���"
s5 | String.Format | 127 | No | "1234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567"
**s6** | String.Format | 128 | **Yes** | "0_���"

Also, the datatype of the variable used in the concatenation doesn't seem to make any difference.  I tried both _string_ and _int_ with the same outcome.

Arke64, 2018-01-31:
We have been able to reproduce the issue using the various Visual Studio inspection tools, but it looks like the variable itself is fine.

However, at least in our tests, when printing out the variable, the contents is correct. Do you see the same behavior?

```cs
Debug.WriteLine(s0);
Debug.WriteLine(s1);
Debug.WriteLine(s2);
Debug.WriteLine(s3);
Debug.WriteLine(s4);
Debug.WriteLine(s5);
Debug.WriteLine(s6);
```

rickmas, 2018-01-31:
Yes, Debug.Writeline() does show the variables properly.  I found this issue when throwing an exception with a string built through concatenation and the output was garbled there too.
```c#
throw new ApplicationException(s6);
```

Arke64, 2018-01-31:
For the 0.8 release, we have fixed the string getting corrupt in display. However, it'll be limited to a max of 128 characters displayed. In a future release we will come back and lift that limit.

closed 2018-01-31 by Arke64

rickmas, 2018-01-31:
Sounds good, thanks!