ghi-electronics/TinyCLR-Ports #174: Enum.ToString() returns value instead of name

rickmas, 2018-02-02:
TinyCLR 0.7.0

How come when I use ToString() on an Enum, I get the value instead of the name?

For example,
```c#
    class Program
    {
        enum Palette : uint
        {
            Cyan = 0xFFFF00,
            Gray = 0x808080,
        };

        static void Main(string[] args)
        {
            Palette p1 = Palette.Cyan;

            Debug.WriteLine($"My Color is {p1.ToString()}");
        }
    }
```
in a console app, I get "My Color is Cyan"

but on my FEZSpider, I get "My Color is 16776960"

Is this expected or a bug?

Palomino34, 2018-02-02:
Try to use: 

`nameof(Palette .Cyan);`

Arke64, 2018-02-02:
It is expected. NETMF, which TinyCLR forked from, never supported this. We haven't added it ourselves yet. Unfortunately the metadata needed to get the names isn't present on the device. To what @Palomino34 suggested, while not ideal, something like the below may work:

```cs
enum Palette : uint {
    Cyan = 0xFFFF00,
    Gray = 0x808080,
}

string GetName(Palette value) {
    switch (value) {
        case Palette.Cyan: return nameof(Palette.Cyan);
        case Palette.Gray: return nameof(Palette.Gray);
        default: throw new ArgumentException();
    }
}
```

rickmas, 2018-02-02:
@Palomino34 thanks for the tip, that was better than what I had.
@Arke64 thanks for the code and for taking the time to provide an explanation.  works great!

closed 2018-02-02 by rickmas