ghi-electronics/TinyCLR-Ports #177: Exception filter causes lockup when exception thrown from instance method

rickmas, 2018-02-16:
TinyCLR 0.8.0
VS2017
FEZ Spider (EMX)

In the following contrived example, when the **catch (...) when (...)** expression evaluates to true the system locks up and control never falls into the catch block. I'm forced to pull power from my board and erase/reload firmware every time.

```c#
using System;

namespace MaschMaker
{
    public class TestANE
    {
        public void ThrowANE() { throw new ArgumentNullException(paramName: "ANE", message: "some message"); }
    }

    class Program
    {
        static void Main()
        {
            try
            {
                new TestANE().ThrowANE();
            }
            catch (ArgumentNullException e) when ("ANE" == e.ParamName)
            { }
        }
    }
}
```

Removing the **when ()** expression makes it work just fine

```c#
            catch (ArgumentNullException e) //when ("ANE" == e.ParamName)
```

**OR** 

keeping the **when ()** expression but changing **ThrowANE()** to a static method also works, probably because **Main()** is static.

```c#
using System;

namespace MaschMaker
{
    public class TestANE
    {
        public static void ThrowANE() { throw new ArgumentNullException(paramName: "ANE", message: "some message"); }
    }

    class Program
    {
        static void Main()
        {
            try
            {
                //new TestANE().ThrowANE();
                TestANE.ThrowANE();
            }
            catch (ArgumentNullException e) when ("ANE" == e.ParamName)
            { }
        }
    }
}
```

**OR** 

keeping the when () expression but moving **ThrowANE()** to be a local method works. This is probably the same case as using a static method but was another test I did.

```c#
using System;

namespace MaschMaker
{
    class Program
    {

        static void Main()
        {
            void ThrowANE() { throw new ArgumentNullException(paramName: "ANE", message: "some message"); }

            try
            {
                //new TestANE().ThrowANE();
                //TestANE.ThrowANE();
                ThrowANE();
            }
            catch (ArgumentNullException e) when ("ANE" == e.ParamName)
            { }
        }
    }
}
```

I obviously have a workaround but would like to use the exception filters if possible.

Arke64, 2018-02-19:
We were able to reproduce it, and not just on the EMX. The generated IL is nearly identical. So it looks to be a problem in the runtime.

Arke64, 2018-08-07:
We've done a bit more investigation and it seems the problem occurs because the handler is too close to the end of the method, which triggers an assert validating bytecode boundaries. For example, blinking an LED after the `try/catch` was enough to let it work. That's also why you could rearrange things slightly to get it to work. We don't yet have an actual fix but we are continuing to investigate.