ghi-electronics/TinyCLR-Ports #188: Need Thread.Interrupt() capability

rickmas, 2018-02-28:
Hello,

Using TinyCLR v0.8.0

Can somebody please give me some insight as to why the **Interrupt()** method doesn't exist for Thread objects in c#?

If I put a thread to sleep using **Timeout.Infinite**, the documentation says it can be awoken using the **Abort()** or **Interrupt()** methods.  The **Abort()** method exists and I see that I could potentially use this to awaken a thread without killing it but would need the **ResetAbort()** method which I also don't see exists.

This leaves the **Suspend()** and **Resume()** methods but the documentation says these are obsolete.

So just curious how to awaken a thread put to sleep indefinitely?
And, are there plans to provide **Interrupt()** and/or **ResetAbort()**?

Thanks!

Arke64, 2018-02-28:
Well, the short answer is we haven't reviewed adding either `Interrupt` or `ResetAbort` yet. In general, I wouldn't rely on the regular .NET API docs too heavily for specifics yet.

In the mean time, if the thread is sleeping indefinitely, you may be able to create a solution using `AutoResetEvent` or `ManualResetEvent` where you call `Wait` instead of `Thread.Sleep`.

rickmas, 2018-03-01:
Sounds good, I look into your suggestions.  Thanks!

closed 2018-03-01 by rickmas

rickmas, 2018-03-05:
@Arke64 , Using ManualResetEvent and WaitOne did the trick.  Thanks for your help!

Arke64, 2018-03-06:
Happy to hear you got it working!