ghi-electronics/TinyCLR-Ports #234: InvalidOperationException thrown trying to Join worker thread from GC thread

rickmas, 2018-03-17:
TinyCLR 0.9.0
VS2017
FEZ Spider (EMX)

I have a process that won't terminate because I have a worker thread running indefinitely.  This would typically be solved by setting the IsBackground property on the worker thread to true, but that property doesn't exist for TinyCLR.  So, I thought I could use a class destructor as an opportunity to stop the worker thread.  This works fine in a desktop app but in TinyCLR I get System.InvalidOperationException.

Here's a basic example showing the issue:

```c#
using System;
using System.Threading;
using System.Diagnostics;

namespace MaschMaker
{
    class Program
    {
        static void Main()
        {
            Worker worker = new Worker();

            //worker.Shutdown();    // works

            // force GC
            worker = null;  
            GC.Collect();
            Debug.WriteLine("Exit Main()");
        }
    }

    class Worker
    {
        private Thread wt = new Thread(() => { Thread.Sleep(1000); Debug.WriteLine("- work is done"); });

        public Worker() => wt.Start();
        ~Worker() => Shutdown();    // does not work

        public void Shutdown()
        {
            Debug.WriteLine("> Wait for work to be done.");
            wt.Join();
            Debug.WriteLine("> Done waiting");
        }
    }
}
```
The exception happens on **``wt.Join()``** in the **``Shutdown()``** method.  Calling Shutdown() from Main(), instead of the destructor, works just fine.

I'm not sure that what I'm doing is the best way to get the worker thread to stop but I wasn't expecting InvalidOperationException either.

Arke64, 2018-03-19:
We can reproduce the exception. In fact, it appears to be a deliberate design choice inside the code. Given that, it's not something we'd remove this late in the release cycle since it may break assumptions elsewhere.

That said, relying on a finalizer to join a thread is inherently brittle given the nature of finalizers. They are not guaranteed to run at all and will not trigger if you have a reference to the object mistakenly held somewhere else. They also may not fire at a deterministic time.

A likely better solution is to use the `Dispose`/`using` pattern (which you already got close to by calling `Shutdown` in `Main`). Additionally, such a method can be used to explicitly stop the thread, using some flag or condition to get the worker thread to exit. The example posted only "works" because the worker thread exits on its own after a brief delay. If it loops, for example, even the desktop pauses in the finalizer thread on the call to `Join`. Since both threads are alive, the process doesn't exit (though the runtime will eventually kill the finalizer thread).

rickmas, 2018-03-20:
The funny thing is that after much refactoring and many layers, I didn't know I was trying to stop a worker thread from a destructor so this exception saved me from a bad design and potential bugs down the road.  Thanks again for your comments/feedback.

-Rick

closed 2018-03-20 by rickmas