ghi-electronics/TinyCLR-Ports #355: TimeSpan.ToString() throws CLR_E_UNSUPPORTED_INSTRUCTION

MikroBusNet, 2018-07-24:
When using the following syntax, you get that exception :
```
var duration = DateTime.UtcNow.Subtract(deb);
var str = duration.ToString();
```
However, the following syntax does not throw the exception :
```
var duration = DateTime.UtcNow.Subtract(deb);
var str = "" + duration;
```

Arke64, 2018-07-24:
This is indeed something we know about. In fact, we deliberately throw that exception because the instruction is indeed not currently supported (it never was). Previously it was just ignored which can cause unexpected behavior. We're still deciding on what we want to do for v1.0 here.

Arke64, 2018-07-24:
Of course the end goal is to actually implement and fully support the instruction. It's the `.constrained` prefix for those curious.

closed 2018-07-24 by Arke64