ghi-electronics/TinyCLR-Ports #368: Add time to event that posted from native to managed.

Palomino34, 2018-07-31:
Example Gpio

interopManager->RaiseEvent(interopManager, "GHIElectronics.TinyCLR.NativeEventNames.Gpio.PinChanged", self->ApiInfo->Name, (uint64_t)pin, (uint64_t)(edge == TinyCLR_Gpio_PinChangeEdge::RisingEdge), 0, 0);

The last param should be time, temporary set to zero.

Same issue for Uart, Can.

Palomino34, 2018-08-07:
Managed driver ignore time, aslo in native, adding more functions to get time for CAN, GPIO, UART... will increase size.

Arke64, 2018-08-07:
We don't currently use it in our handlers but the user might. Either way we can push back.

Palomino34, 2018-08-07:
Just rework, for an event take around 60 bytes. We have Uart x 3 events + Can x 2 events + gpio x 1 event = 360 bytes more.