ghi-electronics/TinyCLR-Ports #520: G120E: "Cannot find any entrypoint"

Palomino34, 2018-10-24:
Not found on G120 but easy to be happened on G120E

[image]

Palomino34, 2018-10-30:
To reproduce issue, easy way is tied pin P3_30 to GND, accessing SDRAM seems failed that cause CRC wrong.

Another way to reproduce by code is:
```
static void Main()
        {
            var p3_30 = GpioController.GetDefault().OpenPin(3 * 32 + 30);

            p3_30.SetDriveMode(GpioPinDriveMode.InputPullDown);
  
            var counter = 0U;
            while (true)
            {
                Thread.Sleep(500);

                System.Diagnostics.Debug.WriteLine("System is running...." + counter++);
            }

        }
```

Arke64, 2018-11-14:
While it doesn't take care of the root issue, we probably want to change all GPIO on all devices to be no-init. The processor defaults should be enough.

Arke64, 2018-11-19:
Folded into #564

closed 2018-11-19 by Arke64