ghi-electronics/TinyCLR-Ports #526: Signal Generator value is incorrect some cases.

Palomino34, 2018-10-26:
```var buffer = new[] {
            TimeSpan.FromMilliseconds(2),
            TimeSpan.FromMilliseconds(3),
            TimeSpan.FromMilliseconds(4),
            TimeSpan.FromMilliseconds(5),
            TimeSpan.FromMilliseconds(6),
            TimeSpan.FromMilliseconds(7),
        };
```

give values 2400, 2800, 4000, 5200, 6000, 6800, on scope
But 

 ```var buffer = new[] {
            TimeSpan.FromMilliseconds(2),
            TimeSpan.FromMilliseconds(2),
            TimeSpan.FromMilliseconds(2),
            TimeSpan.FromMilliseconds(2),
            TimeSpan.FromMilliseconds(2),
            TimeSpan.FromMilliseconds(2),
        };
```

scope shows correctly.

Palomino34, 2018-10-31:
The issue now only found on G120 and G400.
With the zip attach file, G400 will work more accurate, but won't apply to official build for
[AT91SAM9X35_ReWriteILoopDelay.zip](https://github.com/ghi-electronics/TinyCLR-Ports/files/2536366/AT91SAM9X35_ReWriteILoopDelay.zip)
 now.

Palomino34, 2018-11-06:
Close because comment above.

Arke64, 2018-11-07:
We will look into a fix, just not for preview 3.

Palomino34, 2018-11-29:
In Preview3: Not found on G120E, but found on all STM and G400

commit a117e594 (2019-01-02): Fixed #526. Time only wrong on G400, get value closer to right one.
commit a84f6fd4 (2019-01-02): Merge pull request #598 from ghi-electronics/#526_FixedSignalGeneratorIncorrectValue

Fixed #526. Time only wrong on G400, get value closer to right one.
closed 2019-01-02 by Arke64