ghi-electronics/TinyCLR-Ports #631: GpioPin.Read().ToString()

Se3ker, 2019-06-15:
Please implement .ToString() of GpioPin.Read()

Arke64, 2019-06-18:
What do you expect it to return? https://docs.microsoft.com/en-us/uwp/api/windows.devices.gpio.gpiopin doesn't document ToString being overriden.

Se3ker, 2019-06-19:
Debug.WriteLine(GpioPin.Read().ToString());
Either "High/Low" or "1/0" if it has a .ToString() it should implement it.

Its better then:
if (GpioPin.Read() == GpioPinValue.High) Debug.WriteLine("HIGH") else Debug.WriteLine("LOW")

Arke64, 2019-06-19:
Read() just returns a bool, we can't change how ToString() works on that, it'll return "True" or "False".

So `Debug.WriteLine(GpioPin.Read().ToString());` will show "True" or "False", which means "High" and "Low" respectively.

Se3ker, 2019-06-19:
Read() return a GpioPinValue not bool

Arke64, 2019-06-19:
Ah, right. Sorry.

In that case, `GpioPinValue` is an enum, we don't have control over what its `ToString` returns.

More generally in TinyCLR, enums return the underlying integer value in their `ToString`, not their name like on other platforms. While adding that is something on our todo list, it's not an easy fix because the metadata is not on the device. That feature is tracked by ghi-electronics/TinyCLR-Core#419, so we'll close this as a duplicate of that.

However, `GpioPinValue.Low` is 0 and `GpioPinValue.High` is 1, so with the above enum behavior, it's close to what you proposed.

closed 2019-06-19 by Arke64